gcobos/grblesp #7: steppers pause when send gcode via websocket

MillerRen, 2021-03-22:
hi @gcobos thank you  create this great object. It works fine by usb serial but websocket. Steppers always paused and run agin later. I gess maybe the timer0 issue. Timer0 used by wifi feature,but steppers pluse used timer0 too. Have you faced this problem?

gcobos, 2021-03-23:
Hi! Thanks for trying this out. The support for websockets is not fully tested. What are you using as the client for websocket?

MillerRen, 2021-03-24:
lasergrbl

gcobos, 2021-03-26:
Thanks, I'll check it out on the weekend

gcobos, 2021-03-28:
I would recommend to not to use this project with lasergrbl. Many of the fuctionalities of GRBL are not enabled, and only the motor control is ensured to work. I would choose something like https://github.com/bdring/Grbl_Esp32 which has much more support, but you will then need an ESP32, instead of a ESP8266

MillerRen, 2021-03-30:
Thank you for your advice. I am porting this to a pen plotter or laser engraver. Esp8266 has enough pins. It's a nice things that a 2 axis cnc powered by esp8266 without external ic. Because esp8266 is cheap and simple :-)

closed 2021-03-30 by MillerRen