gcobos/grblesp #10: hi , Mr gcobos,can you help me ,anyway thank you

mvwtest, 2021-06-18:
in platformIO ,i buildin it,  
this error like loved me, but i don`t.
please get me a  way, thank U.

=================================================================================
*** [.pio\build\esp12e\lib8e3\grbl\nuts_bolts.cpp.o] Error 1
================================== [FAILED] Took 120.05 seconds ==================================The terminal process "C:\Users\M\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
=================================================================================

mvwtest, 2021-06-18:
and  Vscodium + platform on Win10,
or linux suport only

gcobos, 2021-06-19:
Googling around I found this. https://stackoverflow.com/questions/47946868/how-to-get-rid-of-terminal-will-be-reused-by-tasks-press-any-key-to-close-it

mvwtest, 2021-06-20:
hi gcobos 
thank you commented.
my ? is :

[.pio\build\esp12e\lib8e3\grbl\nuts_bolts.cpp.o] Error 1

not orther,thanks again

gcobos, 2022-01-13:
This is now fixed in main branch. Closing.

closed 2022-01-13 by gcobos