gcobos/grblesp #19: error help

sasan18s, 2025-04-13:
hi . iam test  it with platform io and arduino ide and in theme after verify got error : 
platform io:
[{
	"resource": "/c:/Users/Sasan/platformgit/grblesp/lib/grbl/src/nuts_bolts.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "expected unqualified-id before '(' token",
	"startLineNumber": 57,
	"startColumn": 18,
	"endLineNumber": 57,
	"endColumn": 18
}]
arduino ide : 
expected unqualified-id before '(' token
for line : 
#define max(a,b) (((a) > (b)) ? (a) : (b))
#define min(a,b) (((a) < (b)) ? (a) : (b))
help please 
iam just want control two XY stepper for diy Laser machine and think Maybe It work .

sasan18s, 2025-04-16:
؟؟

gcobos, 2025-04-16:
Hi! It's probably because "min" and "max" is already defined somewhere else. Did you tried just commenting those two lines?

// #define max(a,b) (((a) > (b)) ? (a) : (b))
// #define min(a,b) (((a) < (b)) ? (a) : (b))

sasan18s, 2025-04-17:
yeah I test It and the it error in 
Cpu_map File in : 

  #define CONTROL_INPUT_GPIO_PIN    D3    // GPIO_0 = D3

  #define LIMIT_INPUT_GPIO_PIN    D4    // GPIO_2 = D4

error : 
 
'D3' was not declared in this scope
'D4' was not declared in this scope

and I just turn D3 to 3 - D4 to 4 like that : 

  #define CONTROL_INPUT_GPIO_PIN    3    // GPIO_0 = D3

  #define LIMIT_INPUT_GPIO_PIN    4    // GPIO_2 = D4

 and it Errored : 

Error compiling for board Generic ESP8266 Module.

and if i comment this lines too : 
//  #define CONTROL_INPUT_GPIO_PIN    D3    // GPIO_0 = D3

//  #define LIMIT_INPUT_GPIO_PIN    D4    // GPIO_2 = D4 

it error : 

'LIMIT_INPUT_GPIO_PIN' was not declared in this scope
 
for line : 

        if (!GPIP(LIMIT_INPUT_GPIO_PIN)) { 
in limits.cpp ...
 
what should i do ?

sasan18s, 2025-04-17:
i think esp dip name should be like that : 
                  '    static const uint8_t D0   = 16;',
                  '    static const uint8_t D1   = 5;',
                  '    static const uint8_t D2   = 4;',
                  '    static const uint8_t D3   = 0;',
                  '    static const uint8_t D4   = 2;',
                  '    static const uint8_t D5   = 14;',
                  '    static const uint8_t D6   = 12;',
                  '    static const uint8_t D7   = 13;',
                  '    static const uint8_t D8   = 15;',
                  '    static const uint8_t D9   = 3;',
                  '    static const uint8_t D10  = 1;',
 iam find it in boards.txt in esp8266 folder in arduino library .

gcobos, 2025-05-10:
I'm trying myself to compile the whole package, not having much success. There's a mess with Arduino dependencies, and AsyncTCPWebServer libraries, mixing ESP32 and RP240 in the mix. 

I'm trying to find the cleanest way to make it compile, probably requiring to fix versions of packages and libraries to older versions.

Basically, AsyncTCP is focusing on ESP32 and dropping compatibility with ESP8266, so it's going to get more and more difficult.

gcobos, 2025-05-10:
I have updated the project, fixing versions of platform and libraries to working ones.

closed 2025-05-10 by gcobos